MetalWorkerTools/CNCL #2: App stuck in !Unsupported Licence State!

NeilMarley, 2021-03-21:
**Describe the bug**
Have downloaded to trial - on 2 separate devices. Each time get valid 30-day trial  expiry date shown but App will not run beyond help & licence pages..

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 10
 

**Additional context**
Add any other context about the problem here.

HuubBuis, 2021-03-21:
I am not sure about the error, Could you please post a screen shot to clarify?

NeilMarley, 2021-03-21:
[image]

HuubBuis, 2021-03-21:
This is strange, this error should not (can not) appear but is does. 

The account you use on your PC should be the account used for downloading the app.

Please uninstall the app, reinstall the app and than run it using the account you used for installing it.

Let me know if it worked.

HuubBuis, 2021-03-21:
I have checked the Microsoft information portal. I can't find any trial license usage since the last release. Sadly You are the first that reported an issue. I will setup a clean PC environment for debugging.

NeilMarley, 2021-03-21:
Hi Huub,

 

I tried uninstall & a clean re-install on both devices, when logged-in as the Microsoft account holder and the Demo version still didn’t work. Same error as before, shows a valid 30-day trial date  but !Invalid Licence… etc and all of the licence/purchase options greyed-out.

 

I opted to pay for an App Licence + 1 Year (2 separate transactions…?) which now works.

 

It looks like the issue is with the Trial setup…. 

 

Neil

HuubBuis, 2021-03-21:
Neil,
The PC is being installed, this will take some time because it hasn't run for a long time.
I can only debug a trial license for 30 days on a fresh installed PC.  For that the trial license is tested every release by simulating what is not completely the same.

Thank you for buying a license. I will report a soon as I have a trial license running.

Huub

NeilMarley, 2021-03-21:
Hi Huub,

 

No problem, I have enough with the 12 month licence to have a play.

 

Neil

HuubBuis, 2021-03-22:
Neil,
I have found the error. It was a bug introduced by me.
I have released version 2.3.69 with some enhancements that also fixes this issue.

Take care and stay healthy

Huub

HuubBuis, 2021-06-01:
I can’t get a new version released by Microsoft that has a trial option **and** usage license options. To avoid wasting any more time I have decided to drop the one year and permanent usage licenses. All current and new owners of an app license can use the app without limitations.

closed 2021-06-01 by HuubBuis