MetalWorkerTools/CNCL #5: Grbl_L and probe

alanfr, 2022-03-27:
I was hoping to use CNCL with Grbl-L on my CNC Lathe conversion. Wanted to get grbl-L working before testing out CNCL prior to purchase. BUT I cannot get probe to work?? It works fine with standard grbl 1.1. But with grbl-L it does not. I note that the probe pin shares A5 with the spindle sync.. Is there some issue because of this?? I have not yet added in sensors for this sync but I cannot see why that should prevent Probing?? I realise this is the CNCL log so apologies if this is wrong place.

HuubBuis, 2022-03-28:
Alan,
I use grbl-l on my small (demonstration) lathe and have no problems using probing. I use NO switches and the latest grbl-l version.

Because the probe pin shares the spindle sync pin, you need to **use NO switches, parallel connected**.

If the sync pin is triggered, the probing signal is also active. If that is the case, You have to turn the spindle a bit to release the sync pin signal. **This means the spindle should be stopped during probing**.

CNCL checks if the sync/probe pin is triggered before probing and will give a message if so.

Probing was deactivated in earlier grbl-l versions. **Use the latest grbl-l release** to be sure.

Huub

alanfr, 2022-03-28:
Doh!!! Thanks HuubBois, I had the 2021 version installed. Just updated and it now seems to work OK! Next step is to to add spindle sensors. Then I will try out CNCL. This lathe is a Unimat 3 manual lathe that I am converting to CNC (a learning exercise). I also have a (non working) Emco 5 CNC that I hope to update so that it uses modern electronics and steppers.

HuubBuis, 2022-03-28:
Alan,
Great to hear it is working, have fun,
Huub

HuubBuis, 2022-03-29:
I have ported [grblHAL](https://github.com/grblHAL) to the SAMD21 AND STM32. The main reason for this was to improve the  threading capabilities. I have bench tested and the more complicated calculations improved the threading synchronization. I have implemented them in the latest grbl-l and grbl-l-mega version. These processors are still fast enough for this modification.
The grbl-l version (UNO) has now reached its memory limit. The grbl-l-mega version (MEGA) has plenty of space for improvements. However, for the lathe, I do not think there is much missing in the current versions.
I use WiFi and Bluetooth to connect my lathe and rotary table. There is a grbl port for the ESP32 [Grbl_Esp32](https://github.com/bdring/Grbl_Esp32) that I use for my rotary table [Grbl_Esp32_R](https://github.com/MetalWorkerTools/Grbl_Esp32_R). I would like to implement threading for the grbl_esp32 version. That is on my (long) todo list.
To make testing all these different boards a bit easier, I am modifying my electronics board on the lathe. This will take some time because there a so much things to do.
For a CNC router, using a faster board is a good idea.  3D milling and printing requires a lot of short Gcodes that makes processing time consuming.
My STM32 and SAMD21 port are not suitable to use. They where only ported (quick and dirty) for testing threading.synchronization.
I have bench tested grblHAL and CNCL, this combination works OK.

alanfr, 2022-03-31:
Wow, you have been busy. I had also contacted bdring who did grbl_ESP32 conversion (now called FLUIDNC). He also has a lot to do. Not sure if I could assist in any way; I can program and I have an ESP32Duino to test with. I confess that I am not wholly conversant with the structure used by the likes of Grbl and Marlin (I have only made usual types of changes in configs or map.h; never explored other aspects); but happy to give it a go. Any pointers (eg list changes made to standard grbl to add in threading).

HuubBuis, 2022-03-31:
Alan,
The ESP32 programming is quit difficult compared to the Atmega 328P. Without a decent debugger, programming is very time consuming independent of the processor type.
 I use Atmel studio for  grbl-l and grbl-l-mega and an Atmel dragon to debug but this is quite expensive.
For the SAMD21 I use a segger jlink edu mini that is cheap.
For the STM32, a ST-link, that is the cheapest debugger, only a few dollars.
As far as I know, the ESP32 could be debugged using jlink.

Not all boards have a debug connector. I do not use boards without a debug connector. Most ESP32 boards don't have a debug connector. There should be a ESP32 Wrover kit that has a built in debugger. 
For the ESP32, I will make a debug shield.

FluidNC uses Visual Studio Code and Platform IO as development platform. Not really my first choice, but very better than the Arduino IDE. FluidNC is a total redesign of the original grbl code. Grbl had to deal with lack of program space and readabilty was not a priority. 
Until now, I am not able to compile FluidNC. I will give it try when FluidNC is out of beta and I have finished the new electronics of my lathe. If I am going to implement threading it will be in the FluidNC version. At that time I will contact BDring what the requirements are so that it will be added to his version. I really want to avoid making my own lathe version. To be honest, I don't think I will start this project within a year.

> Not sure if I could assist in any way; I can program and I have an ESP32Duino to test with
Any pointers (eg list changes made to standard grbl to add in threading)

There is a list of changes made to grbl for implementing threading. 
[https://github.com/MetalWorkerTools/grbl-L-Mega/wiki/Threading-Code-explanation](url)
But that is just a summary of how it works and what files are changed, not the changes.

The todo list for grbl-l is empty.
For the grbl-l-mega I will add code so that the probe and synchronization each get their own pin. I also would like to add the response to buttons like I implemented in the grbl_esp32_r version. But this has no priority at the moment.

HuubBuis, 2022-03-31:
Alan,

I tried to build the FLuidNC code and it didn't (as expected). I checked for updates and updated the Espressif 32 library to version 3.5
Now I can build FluidNC.

Huub

wandlerdarwin, 2023-08-10:
I would like to get my hands on your STM code if you have it available for the STM32F103C8T6 I plug this board into my 4 axis daughter board running SERVO Motors with 8000 PPR for extreme accuracy and control with super speed on my 5 foot x 10 foot CNC routers, plasma cutters and laser engravers wandler@abnet.ca

HuubBuis, 2023-08-10:
I have checked my code, but can't find any grbl/grblHAL code for any STM32 board. I have tested my own threading implementation on a grblHAL / STM32 Nucleo F411RE but it seems I have deleted that code also. 
The only 32 bit processor code I have found is grblHAL ported to a SAMD21 board that I also have used to test different threading implementations. This code won't run any other program because the implementation was pretty dirty. [SAMD21](https://github.com/HuubBuis/grblHAL).
You could port a current grblHAL version to run on your board.
All threading code if have written is implemented using a low resolution (4..16 PPR) encoder (magnets and proxy) that is serviced by software interrupts.
For your 8000 PPR encoder you need a hardware encoder input. Some STM32 processors can do that job.

wandlerdarwin, 2023-08-10:
Probe pin fix #32

https://github.com/usbcnc/grbl/pull/32/commits/222e043b384b27504329b8e7c658f2d2b8033f31

closed 2023-08-14 by HuubBuis

HuubBuis, 2023-08-17:
> I would like to get my hands on your STM code if you have it available for the STM32F103C8T6 I plug this board into my 4 axis daughter board running SERVO Motors with 8000 PPR for extreme accuracy and control with super speed on my 5 foot x 10 foot CNC routers, plasma cutters and laser engravers [wandler@abnet.ca](mailto:wandler@abnet.ca)

I have found my [STM32 grblHAL port for STM32F4xx  ](https://github.com/HuubBuis/grblHAL) for testing threading spindle synchronization.