MetalWorkerTools/CNCL #6: Error setting workposition

Frenergy, 2022-12-20:
Hello,

I've uploaded GRBL-L to my Arduino Uno (no things yet connected, nor steppers, nor limit switches, etc). When connecting the Arduino in CNCL I get always the same bug:

Error Setting Workposition
Com Error Setting Workposition <X>
Check Your Positions
Controller will be disconnected

In GRBL info I saw this possible cause:
Last Error Command: G92X0 **_,_** 0000
Respons: error:1\r\n

Is it possible that the error is caused by the use of a comma instead of a period?

I've tried mulitple times, this error keeps occuring on all of my arduino's (also MEGA). I'm using Windows 11. Connection with UGS seems to be no problem.

[image]

MetalWorkerTools, 2022-12-20:
Hello,I do not know if the dot/comma is a problem. I ca check this tomorrow.In your grbl info screen, I see a lot of com errors. I have had such a problem using WiFi and a (expensive) TP-Link router. I found that this router even timed out when sending a ping to a local IP-Adress. You could try connecting using USB to see if the problem persists.Huub In the grbl info screen, I see 70 co

Frenergy, 2022-12-21:
Hello Huub,

I've connected the arduino only via USB. The problem keeps occuring even when my internet connection is turned off. The amount of communication errors is due my endless attempt of trying to connect the arduino.

Looking forward to your investigation of the possible cause coming from the comma/period

Frédéric

HuubBuis, 2022-12-21:
I am puzzled why I have no problems on my PC(s). The PC is configured in the "nl-NL" language style and the decimal separator defaults to a comma (,). Whatever I change my Windows10 language and number notation to, it formats using de dot (.) as decimal separator. So I can't reproduce this error for testing.

I have changed the code so that (Gcode) numbers are formatted in the "nl-NL" language style.  In this style, I have changed the decimal separator to a dot (.)
I will upload this new version later this night after the  release documentation is done.

Frenergy, 2022-12-22:
Hi Huub,

I've tried your suggestion and was able to make connect the arduino! Looking forward to test your software :)

HuubBuis, 2022-12-22:
Hi Frédéric,

Great to hear you have things working for you.
I have uploaded a new CNCL & CNCR version to the Microsoft store. Both versions also have some enhancements. Once Microsoft has completed the certification, I will let you know.

regards, Huub

HuubBuis, 2022-12-22:
The new release is approved by Microsoft. It should be visible in the Microsoft store in a few hours. You can initiate an update using the Microsoft store to speed things up.

closed 2023-08-14 by HuubBuis