MetalWorkerTools/CNCL #7: ThreadingSS questions

RogEnk, 2023-01-03:
Hi Huub, really enjoying CNCL and now working on getting Spindle Sync working and have a couple of questions that the answers may help others:
Using an Arduino UnoMega 2560 with the lates edge release, and 2 hallsensors identical to yours.
1. I have both index and pulse magnets, both hall sensors light up when magnet (4mm neodymium’s) move past them - I opted for 8 magnets for sync and 1 for index (believing only 1 is required). What should the settings be for $40 and $41? I have tried: 3 for $40 and 10 for $41, and I cannot get sync to work consistently, when it did work every 5th or so pass went out of sync and feed rate shot up destroying what was a nice thread on my test piece.
2. Is 8 magnets too much for the Mega by producing too many pulses? Lowest RPM I can do in low gear on my lathe is around 50-60 RPM.

Once again: thanks for producing this!

MetalWorkerTools, 2023-01-03:
Hi,
I use 8 magnets on the mini lathe (UNO) and 4 magnets on my bigger lathe (MEGA). I do threading at 300 RPM and have tested 500 RPM on the mini lathe. So I don't expect any troubles using 8 magnets at 50 RPM.  

Start by using only an Index pulse. Set $40 to 1, $41 to 25.  
Thread a 1 mm pitch, 50 mm long and and use a cutting depth of 0 mm. Don't place a tool, **just cut air**.  Set your Home Position Z to 50 mm so that the stepper has plenty of time to synchronize. You should listen to the Z-axis stepper. It should run at a constant speed. Also monitor the Synchronization Error. It should be close to 0 once synchronized and it should be synchronized before you reach position Z=0.  
You can try to slow down the spindle by hand. You should see the Synchronization Error go up and going back to 0. This must go flawless, if so you can repeat the test at 75, 100, 150, 200, 250 and 300 RPM.  
If this goes well, you can do cut an actual thread 0.1 mm deep.  Repeat threading 50 times (at 0.1 mm total depth) and check the thread. If you miss an index (synchronization) pulse or detect one to many, you will see it on the thread. You should listen to the Z-axis stepper. It should run at a constant speed.
If this goes well, you can cut a thread at it's full depth using small steps. The spindle must keep running close to the requested speed.
The next step is adding synchronization pulses and repeat the above tests. You have 8 magnets so $40 should be set to 8.

The Index pulse should be **between** 2 synchronization pulses. It should not coincident with a synchronization pulse.

If you monitor the RPM on the CNCL screen, it should be pretty stable and **match the actual spindle RPM**.

$41 is a noise filter. It can reduce/prevent false index and synchronization pulses. It is only available on the Mega. 25 is a good starting value. You can increase this value if your pulse detection isn't stable. If so, double this value and try again. If then, the synchronization is better, you have a problem.  You can increase this value until the synchronization is OK and than set it 20% higher. 

Let me know the results

Regards, Huub

RogEnk, 2023-01-03:
Thanks Huub!
Forgot to ask what the setting for $10 you recommend? CNCL suggests 3, I am getting spindle speed reporting and it appears stable and accurate, sync errors reporting are always 0, but I suspect I may need to modify $10 for better reporting?

I will test the settings you have so far given and report back, really appreciate your help with this.
Forgot to say how much I appreciate your work on the Mega threading price too, it is awesome!

Regards,
Rog

MetalWorkerTools, 2023-01-03:
I have $10 set to 6 to enable reporting of spindle synchronization error.

RogEnk, 2023-01-04:
Thank you once again!Realized I had some EMF issues, so did a complete rewire with shielded cables inside box when I saw the RPM gauge indicating red and 1000+ RPM.Now I can reliably run with just index, max sync error I saw was +/- 002, not triedcutting yet, but I did try with $40 at 8 and again Z axis moves to start position and stops until cycle ends, X retracts and Z clicks or bumps.I am powering the hall sensors via 5v from the Mega, which is powered via USB. Will wire in a 9v supply for Mega tomorrow and test again.Sent from my iPhone

RogEnk, 2023-01-04:
Working perfectly now!
1. I did not understand (and never have to be honest) the bitwise maths.. So knowing $10 should be '6' really helped with reporting and showed a massive misread of spindle sync caused by EMF from motors, sheilding solved that.
2. Analog pin15, I had misread and connected the digital pin!
3. $40 = 8 for number of sync magnets and $41 = 25 worked with 001 - 002 sync errors only at first start as Z stepper reached speed.

Loving this, thank you so much for sharing this with the world, and for helping out a stranger!

Kind regards,
Rog Enkel (Yes I can live up to the Swedish meaning of my lastname lol)

JeffHank1, 2024-05-05:
I am using 32 bit GrblHal control boards for my lathe and CNC machines now. Does attaching the Index and Spindle sync sensor inputs to Y limit and Probe pins still apply when using GrblHal? I can use either a typical motor driven C axis, or Closed loop C stepper. 
The motor will need sensors, but can your CNCL program also thread using a stepper driven C?

HuubBuis, 2024-05-06:
> Does attaching the Index and Spindle sync sensor inputs to Y limit and Probe pins still apply when using GrblHal?

If you use GrblHal, than you have to use the pinout for that board/GrblHal version. It is unlikely that the GrblHal developer has used the Y-limit and Probe pin for index and synchronisation.

> can your CNCL program also thread using a stepper driven C?

CNCL has 2 threading options (Tasks). One uses Spindle Synchronization and G33 gCodes, the other uses the Y-axis for driving the spindle motor. I use both ways, the stepper (Threading task) for more accuracy and the spindle synchronization (ThreadingSS task) for faster threading (my spindle motor has a lot more power than my stepper motor).
CNCL has a one month trial period. Use the trial period to find out how it works on your setup.

HuubBuis, 2024-05-06:
> I can use either a typical motor driven C axis, or Closed loop C stepper.

If you have a step/dir controlled motor on the C-axis, you can also use the C-axis as indexer. Then things like radial and axial pattern drilling, single point knurling, broaching and engraving are possible. That makes the lathe a versatile tool.

I have a Nema24 stepper on the C-axis of my mini lathe and a 600W servo on my bigger lathe.

JeffHank1, 2024-05-06:
Thank you for the reply (on a closed topic, oops). Looks like I will have to revert to a Mega2560 since Tergeio has buried the pin information for now. Presumably he is not yet ready to promote threading, spindle sync, etc. = no problem. I have 3 different 32 bit control boards running GrblHal and prefer them to the old venerable 8 bit Arduino boards. I have a Red Mach3 board, a MKS Tinybee, and a SKR Pro v1.2. Since I have the Mega boards sitting in a drawer I will reprogram one with Grbl L Mega for simplicity. 
Arduino, sigh, Compiling in the IDE, sigh. It’s much simpler with a hex file and X loader ha ha. Cheers.

HuubBuis, 2024-05-06:
I use a MKS Tinybee on my CNC router and I like it very much.
On the lathe, the amount of gcodes lines to send is limited. I experimented with some Arduino Arm and STM boards but could not notice any improvements to the Mega board.
On the Mega and UNO, spindle sync and probe share the same pin. I am going to (try) put them on separate pins because it could interfere with unattended operations. That new pin will probably be the "Cycle Start" pin.  It is on my todo list but it won't be for this year.
Only a few [GrblHal boards support threading](https://github.com/grblHAL/drivers). Most of them are marked as work in progress.

JeffHank1, 2024-05-06:
I noticed the GrblHal 'Work in Progress' grid for most of the 32 bit boards. I agree they are really nice for the CNC. The movements can be much quicker, especially with closed loops steppers. Programing the Mega card was no problem, and now I have an easy to use Hex file too ;)
In CNCL + Gbrl L Mega, are you using the Y as a C axis? I have wired the Y to a C stepper driver for now.
I did not see any mention of a Rotary axis in the cpu_map file for Grbl L Mega. I have my lathe moving correctly in X and Z after rewiring for the right pins, but still no C action. I haven't yet found the setting for Spindle mode since I just got it all wired and moving.

HuubBuis, 2024-05-06:
> In CNCL + Gbrl L Mega, are you using the Y as a C axis? 

Yes, the Y-axis is used to drive a C-axis. Grbl does not support a C-axis. CNCL does the math to convert rotation to stepper pulses. The resolution of the Grbl Y-axis (steps/mm) has to be set to steps needed to rotate the C-axis 1° (steps/°).

In the WiKi you find the information to [setup GRBL-L-Mega](https://github.com/MetalWorkerTools/grbl-L-Mega/wiki/Threading-setup-and-use)