MetalWorkerTools/CNCL #8: Reference not working anymore

Fietje10, 2023-08-12:
Hi, 

since I have updated to 2.3.87 the reference buttons in Setup Menu won't work anymore. I am using the latest version of GRBL-L Mega which is regularly connected via USB. Before the update everything was working and even now a test with UGS is homing just fine.

There are no error messages, after pressing either "Ref X", "Ref Z" oder "Ref XZ" the buttons become inoperational for a short period, but no movement of any axis is trigered. It is quite less fun working without referencing on the lathe.

HuubBuis, 2023-08-12:
@Fietje10 
This is a bug in release 2.3.87. It is fixed in release 2.3.89 (05-2023). 
If you download this (latest) release all should be working again.
I am sorry for the inconvenience.
Regards, Huub

Fietje10, 2023-08-13:
Thanks for the quick response Huub, seems to be that I should have checked for the most recent release first... sorry about that.

closed 2023-08-13 by Fietje10

HuubBuis, 2023-08-13:
@Fietje10
Great to hear all is working again.
Have fun,
Regards, Huub

Fietje10, 2023-08-13:
Hi Huub, I was just checking it out actually at my workshop and it seems to be the same even with the newest release . Made sure I actually have the latest release by uninstalling and installing it again, but the issue remains. Is there anything you can imagine I could check ?

HuubBuis, 2023-08-13:
@Fietje10 
- Check the CNCL version in the Info | License Info screen. It should show 2.3.89
- Check the grbl-l version in the Info | Grbl Info screen. It should report GRBL_VERSION_BUILD "20190917"]
- I can't check the build information at the moment. I will do that in a few hours.
- Check the information in the Info | Grbl Info screen after referencing a single axis to see what errors are reported.
- Move (jog) to a reference position to check the switch is activated **and** seen by Grbl-L. In the Info | Grbl Info screen you can see the activation by a switch (Last Machine Status). J

Regards, Huub

Fietje10, 2023-08-13:
Thanks for your help !
- CNCL version is 2.3.89.0 freshly installed today from Microsoft store
- grbl-l version seems to be more recent, 20211002 (see screenshot)
- There are no errors reported after pressing "Ref X" in GRBL Info (screenshots before and after)
- Switches on both axis are properly recognized by GRBL, they also work fine if hard limits are enables as limits.

[image]
[image]

HuubBuis, 2023-08-13:
@Fietje10 ,
You are using the Grbl-L-Mega version and the latest CNCL version.
I normally run the CNCL develop version (daily) without issues. I am going to install the MS store version and will test that version using a Grbl-L-Mega board (my currently connected board).
I will let you know a.s.a.p.
Regards, Huub

HuubBuis, 2023-08-13:
@Fietje10 
I have downloaded the MS store version and I can't reference anymore, strange but true. I will fix that today by publishing a new release. I will let you know when Microsoft has approved the new release and it is available for downloading. 
Regards, Huub

HuubBuis, 2023-08-13:
@Fietje10 ,
The new release 2.3.91 is approved by Microsoft and can be downloaded. However it has the same problem. Seems there is a difference in my local release version and the uploaded one. I have to dig into that problem, that is something new.
Will be continued,
Regards Huub

Fietje10, 2023-08-13:
Sounds like quite hard to debug problem ... sometimes software can be painful. No rush from my side, I am just more or less playing around on my hobby-lathe with it, still have to learn more, but I am really enjoying your software. Thanks !

HuubBuis, 2023-08-13:
@Fietje10 
I have found the problem (disabled code not ready for release yet) and fixed it. You can download version 2.3.93 from the Microsoft store. If no updates are shown, it can be because you already updated today. Than wait for another day or remove the current installation and install it again.
Software like hardware can be troublesome. Nevertheless, it should work. I try to solve the "show killer" bugs a.s.a.p. because they can spoil a nice day.
Thank you for taking the time to report this problem,
Regards, Huub